Inxton/iXlinker #9: No PLC project found in this solution

patopat, 2022-06-09:
Hi,

I have a message "No PLC project found in this solution" when I run the iXlinker from Visual Studio.

Best regards,
Pat

patopat, 2022-06-09:
I disabled the "Independent Project File" option in the PLC and the iXlinker has started.

PTKu, 2022-06-11:
Thanks @patopat looking into it. At this point independent project files are not supported.

TomKovac, 2022-06-15:
closed by #10

TomKovac, 2022-06-15:
Hi @patopat. Try to update to version 0.1.4-alpha.77. Let me know if it works.

patopat, 2022-06-16:
Hi @TomKovac. Now the iXlinker start correctly but when I click on Generate mappings button I have the following error :
[image]

patopat, 2022-06-16:
@TomKovac yes you are right. I have also additional library repositories.

patopat, 2022-06-16:
@TomKovac  I removed all custom library repo and its reference in the project but the problem persist.

TomKovac, 2022-06-16:
Hi @patopat, can you please share the list of Beckhoff libs? I will try it on my side.

patopat, 2022-06-16:
@TomKovac  Here is the list of the lib

[image]

TomKovac, 2022-06-17:
Hi @patopat, I have tried to add the same PLC libraries (just by name as I do not know your exact library versions) and it still works. So I will add some additional diag messages to figure out the source of the trouble. I'll let you know as soon as I'm done.

TomKovac, 2022-06-17:
Hi @patopat, please try to update to version 0.1.4-alpha.79. This could give us more information.

patopat, 2022-06-17:
Hi @TomKovac 
[image]

patopat, 2022-06-17:
@TomKovac 

  <ItemGroup>
    <PlaceholderReference Include="System_VisuElemMeter">
      <DefaultResolution>VisuElemMeter, 3.5.13.0 (System)</DefaultResolution>
      <Namespace>VisuElemMeter</Namespace>
      <SystemLibrary>true</SystemLibrary>
      <ResolverGuid>2717eb6a-dd07-4c66-8d8d-cacebd7b18ae</ResolverGuid>
    </PlaceholderReference>
    <PlaceholderReference Include="System_VisuElems">
      <DefaultResolution>VisuElems, 3.5.13.40 (System)</DefaultResolution>
      <Namespace>VisuElems</Namespace>
      <SystemLibrary>true</SystemLibrary>
      <ResolverGuid>2717eb6a-dd07-4c66-8d8d-cacebd7b18ae</ResolverGuid>
    </PlaceholderReference>
    <PlaceholderReference Include="System_VisuElemsSpecialControls">
      <DefaultResolution>VisuElemsSpecialControls, 3.5.13.0 (System)</DefaultResolution>
      <Namespace>VisuElemsSpecialControls</Namespace>
      <SystemLibrary>true</SystemLibrary>
      <ResolverGuid>2717eb6a-dd07-4c66-8d8d-cacebd7b18ae</ResolverGuid>
    </PlaceholderReference>
    <PlaceholderReference Include="System_VisuElemsWinControls">
      <DefaultResolution>VisuElemsWinControls, 3.5.13.20 (System)</DefaultResolution>
      <Namespace>VisuElemsWinControls</Namespace>
      <SystemLibrary>true</SystemLibrary>
      <ResolverGuid>2717eb6a-dd07-4c66-8d8d-cacebd7b18ae</ResolverGuid>
    </PlaceholderReference>
    <PlaceholderReference Include="System_VisuElemTextEditor">
      <DefaultResolution>VisuElemTextEditor, 3.5.13.0 (System)</DefaultResolution>
      <Namespace>VisuElemTextEditor</Namespace>
      <SystemLibrary>true</SystemLibrary>
      <ResolverGuid>2717eb6a-dd07-4c66-8d8d-cacebd7b18ae</ResolverGuid>
    </PlaceholderReference>
    <PlaceholderReference Include="system_visuinputs">
      <DefaultResolution>visuinputs, 3.5.13.0 (system)</DefaultResolution>
      <Namespace>visuinputs</Namespace>
      <SystemLibrary>true</SystemLibrary>
      <ResolverGuid>2717eb6a-dd07-4c66-8d8d-cacebd7b18ae</ResolverGuid>
    </PlaceholderReference>
    <PlaceholderReference Include="System_VisuNativeControl">
      <DefaultResolution>VisuNativeControl, 3.5.13.0 (System)</DefaultResolution>
      <Namespace>VisuNativeControl</Namespace>
      <SystemLibrary>true</SystemLibrary>
      <ResolverGuid>2717eb6a-dd07-4c66-8d8d-cacebd7b18ae</ResolverGuid>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_Drive">
      <DefaultResolution>Tc2_Drive, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_Drive</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_EtherCAT">
      <DefaultResolution>Tc2_EtherCAT, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_EtherCAT</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_IoFunctions">
      <DefaultResolution>Tc2_IoFunctions, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_IoFunctions</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_MC2">
      <DefaultResolution>Tc2_MC2, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_MC2</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_MC2_Drive">
      <DefaultResolution>Tc2_MC2_Drive, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_MC2_Drive</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_NC">
      <DefaultResolution>Tc2_NC, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_NC</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_Standard">
      <DefaultResolution>Tc2_Standard, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_Standard</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_System">
      <DefaultResolution>Tc2_System, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_System</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc2_Utilities">
      <DefaultResolution>Tc2_Utilities, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc2_Utilities</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="Tc3_Module">
      <DefaultResolution>Tc3_Module, * (Beckhoff Automation GmbH)</DefaultResolution>
      <Namespace>Tc3_Module</Namespace>
    </PlaceholderReference>
    <PlaceholderReference Include="VisuDialogs">
      <Namespace>VisuDialogs</Namespace>
    </PlaceholderReference>
  </ItemGroup>

patopat, 2022-06-17:
@TomKovac I noticed there isn't version to select compared to other libraries
[image]

TomKovac, 2022-06-17:
Hi @patopat. So it seems that the additional diag message tells the truth. As you can see VisuDialogs does not have a defined version. I could add a mechanism to skip such a library. 
Regarding the independent IO it is not supported at the moment. Maybe we will implement it in the future. 
In my opinion, the much more needed functionality to link motion-related terminals as the complete axis.

TomKovac, 2022-06-20:
Hi @patopat, please try this new version 0.1.4-alpha.80 and let me know if the Independent project files for the IO devices work.

patopat, 2022-06-20:
Hi @TomKovac 
```CMD
2022-06-20 09:12:46.421 +02:00 [INF] iXlinker started
2022-06-20 09:12:46.439 +02:00 [INF] with arguments:
2022-06-20 09:12:46.439 +02:00 [INF] -t C:\sw\wks.tc3\xxxxxxxxxx.tsproj -p Debug|TwinCAT RT (x64) -c C:\sw\wks.tc3\xxxxxxxxxxxx.plcproj -g True -d C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\Common7\IDE\devenv.com -n 0
2022-06-20 09:12:47.616 +02:00 [INF] Project contains 3 IoDevices in individual project files
	ETHERCAT_1
	ETHERCAT_2
	Device 1 (USB Remote)

2022-06-20 09:12:47.706 +02:00 [ERR] Unable to discover default resolution for the Plc library: Include: VisuDialogs | Namespace: VisuDialogs | DefaultResolution: ???
Variable: ProjectItemGroupPlaceholderReference.DefaultResolution has an invalid reference!!!
Method:GetPlcLibraries
Object reference not set to an instance of an object.
2022-06-20 09:12:47.913 +02:00 [INF] Reading IO devices in the XAE project: "XaeXXXXXX"!!!
2022-06-20 09:12:48.600 +02:00 [ERR] Not unique box name: null found in: ETHERCAT_1. in the XAE project file: C:\sw\wks.tc3\xxxxxxxxx\_Config\IO\ETHERCAT_1.xti!!!
2022-06-20 09:12:48.601 +02:00 [ERR] Not unique box name: null found in: ETHERCAT_1. in the XAE project file: C:\sw\wks.tc3\xxxxxxxxx\_Config\IO\ETHERCAT_1.xti!!!
2022-06-20 09:12:48.602 +02:00 [ERR] Not unique box name: null found in: ETHERCAT_1. in the XAE project file: C:\sw\wks.tc3\xxxxxxxxx\_Config\IO\ETHERCAT_1.xti!!!
2022-06-20 09:12:48.602 +02:00 [ERR] Not unique box name: null found in: ETHERCAT_1. in the XAE project file: C:\sw\wks.tc3\xxxxxxxxx\_Config\IO\ETHERCAT_1.xti!!!
2022-06-20 09:12:48.603 +02:00 [ERR] Not unique box name: null found in: ETHERCAT_1. in the XAE project file: C:\sw\wks.tc3\xxxxxxxxx\_Config\IO\ETHERCAT_1.xti!!!
2022-06-20 09:12:48.614 +02:00 [INF] Tsproj file contains not unique device name or box name! No outputs generated!!!
```

TomKovac, 2022-06-20:
Hi @patopat, it looks like some boxes under ETHERCAT_1 have not defined the name. And it seems like it is the slaves directly connected to ETHERCAT_1.

patopat, 2022-06-20:
@TomKovac 

It could be the device HD67E19-A1 from ADFweb.com ?

TomKovac, 2022-06-20:
@patopat, thank you very much.
So it seems that you are using independent project files also for the slave devices. I thought it is only possible at the master device level, as I have never used it before. Maybe it would be better to change my settings to match yours, then create the new project from scratch and scan the EtherCAT network. What are your settings under TwinCAT\XAE Enviroment\File settings?
Mine is like this:
[image]

patopat, 2022-06-20:
@TomKovac 
Easy all is set to true :smile:
[image]

TomKovac, 2022-06-21:
Hi @patopat 
I have made some modifications, please update and check the result.

[https://www.nuget.org/packages/Inxton.iXlinker/0.1.4-alpha.81](url)

patopat, 2022-06-21:
@TomKovac  It's better thank you . There some files in the PLC project  but seems there is only the ETHERCAT_1 master.

[image]

```CMD
2022-06-21 17:24:59.176 +02:00 [INF] iXlinker started
2022-06-21 17:24:59.187 +02:00 [INF] with arguments:
2022-06-21 17:24:59.187 +02:00 [INF] -t C:\sw\wks.tc3\xxxxxxxxxxxxxxxx.tsproj -p Debug|TwinCAT RT (x64) -c C:\sw\wks.tc3\xxxxxxxxxxxxxxxx.plcproj -g True -d C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\Common7\IDE\devenv.com -n 2
2022-06-21 17:24:59.861 +02:00 [INF] Project contains 3 IoDevices in individual project files
	ETHERCAT_1
	ETHERCAT_2
	Device 1 (USB Remote)

2022-06-21 17:24:59.909 +02:00 [ERR] Unable to discover default resolution for the Plc library: Include: VisuDialogs | Namespace: VisuDialogs | DefaultResolution: null
Variable: ProjectItemGroupPlaceholderReference.DefaultResolution has an invalid reference!!!
Library is going to be ignored.
Method:GetPlcLibraries
2022-06-21 17:24:59.937 +02:00 [INF] Reading IO devices in the XAE project: "XaeXXXX"!!!
2022-06-21 17:25:00.539 +02:00 [INF] PdoEntryStructures :0
2022-06-21 17:25:00.541 +02:00 [INF] PdoStructures :9
2022-06-21 17:25:00.542 +02:00 [INF] BoxStructures :1
2022-06-21 17:25:00.542 +02:00 [INF] DeviceStructures :3
2022-06-21 17:25:00.542 +02:00 [INF] TopologyStructures :1
2022-06-21 17:25:00.543 +02:00 [INF] Cleaning existing GVL!
2022-06-21 17:25:00.544 +02:00 [INF] Existing GVL cleared!
2022-06-21 17:25:00.545 +02:00 [INF] Cleaning existing DUTs!
2022-06-21 17:25:00.553 +02:00 [INF] Existing DUTs cleared!
2022-06-21 17:25:00.554 +02:00 [INF] Exporting Pdo entry structures to the folder C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\PdoEntries
2022-06-21 17:25:00.555 +02:00 [INF] PDO entry structures exported!!!
2022-06-21 17:25:00.555 +02:00 [INF] Adding PDO entry structures into the PLC project!!!
2022-06-21 17:25:00.556 +02:00 [INF] PDO entry structures added into the PLC project!!!
2022-06-21 17:25:00.556 +02:00 [INF] Exporting Pdo structures to the folder "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\PDOs"
2022-06-21 17:25:00.569 +02:00 [INF] PDO structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\PDOs" !!!
2022-06-21 17:25:00.569 +02:00 [INF] Adding PDO structures into the PLC project!!!
2022-06-21 17:25:00.571 +02:00 [INF] PDO structures added into the PLC project!!!
2022-06-21 17:25:00.572 +02:00 [INF] Exporting Box structures to the folder "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\Boxes"
2022-06-21 17:25:00.574 +02:00 [INF] Box structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\Boxes" !!!
2022-06-21 17:25:00.576 +02:00 [INF] Adding Box structures into the PLC project!!!
2022-06-21 17:25:00.576 +02:00 [INF] Box structures added into the PLC project!!!
2022-06-21 17:25:00.577 +02:00 [INF] Exporting Device structures to the folder "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\Devices"
2022-06-21 17:25:00.579 +02:00 [INF] Device structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\Devices" !!!
2022-06-21 17:25:00.580 +02:00 [INF] Adding Device structures into the PLC project!!!
2022-06-21 17:25:00.581 +02:00 [INF] Device structures added into the PLC project!!!
2022-06-21 17:25:00.582 +02:00 [INF] Exporting Topology structures to the folder "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\Topology"
2022-06-21 17:25:00.583 +02:00 [INF] Topology structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\DUTs\IO\Topology" !!!
2022-06-21 17:25:00.586 +02:00 [INF] Adding Topology structures into the PLC project!!!
2022-06-21 17:25:00.587 +02:00 [INF] Topology structures added into the PLC project!!!
2022-06-21 17:25:00.587 +02:00 [INF] Exporting Base structures to the folder "C:\sw\wks.tc3\WISAMO\xxxxxxxxxxxxxxxx\DUTs\IO\Base"
2022-06-21 17:25:00.591 +02:00 [INF] Base structures exported to the folder "C:\sw\wks.tc3\WISAMO\xxxxxxxxxxxxxxxx\DUTs\IO\Base" !!!
2022-06-21 17:25:00.592 +02:00 [INF] Adding Base structures into the PLC project!!!
2022-06-21 17:25:00.593 +02:00 [INF] Base structures added into the PLC project!!!
2022-06-21 17:25:00.594 +02:00 [INF] Exporting GVL : GVL_iXlinker to the folder C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\GVLs
2022-06-21 17:25:00.597 +02:00 [INF] GVL GVL_iXlinker exported to the folder C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\GVLs
2022-06-21 17:25:00.598 +02:00 [INF] Adding GVL into the PLC project!!!
2022-06-21 17:25:00.599 +02:00 [INF] GVL added into the PLC project!
2022-06-21 17:25:00.600 +02:00 [INF] Writing changes of the PLC project XXXX into the file XXXX.plcproj!!!
2022-06-21 17:25:00.608 +02:00 [INF] Changes of the PLC project XXXX written into the file XXX.plcproj!!!
2022-06-21 17:25:00.610 +02:00 [INF] Starting process Filename: "C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\Common7\IDE\devenv.com", Arguments: ""C:\sw\wks.tc3\xxxxxxxxxxxxxxxx.tsproj" /build "Debug|TwinCAT RT (x64)" /Out "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\\build.txt""
2022-06-21 17:25:00.612 +02:00 [INF] Building XAE project: "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx.tsproj" !!!
2022-06-21 17:29:57.348 +02:00 [INF] XAE project: "C:\sw\wks.tc3\xxxxxxxxxxxxxxxx.tsproj" built
2022-06-21 17:29:57.350 +02:00 [INF]  in 296735 ms!!!
2022-06-21 17:29:57.351 +02:00 [INF] Generating mappings into the project file C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\_Config\PLC\XXXX.xti!!!
2022-06-21 17:29:57.417 +02:00 [INF] Mappings generated into the project file C:\sw\wks.tc3\xxxxxxxxxxxxxxxx\_Config\PLC\XXXX.xti!!!
2022-06-21 17:29:57.419 +02:00 [INF] Complete process 298230 ms!!!
```

TomKovac, 2022-06-22:
Hi @patopat, it looks like ETHERCAT_1 has no slaves, ETHERCAT_1  is disabled, or all salves of the ETHERCAT_1 are disabled. To filter out the last two possibilities, please run again iXlinker, but uncheck this:
[image]

patopat, 2022-06-22:
@TomKovac, Nothing is disabled and there are slaves.

[image]

Effectively, with option unchecked the ETHERCAT_1 appears, strange...
[image]

TomKovac, 2022-06-22:
Hi @patopat, I will check it out. Maybe the information if the slave is disabled is stored differently in the case of independent project files used. I found out that's the case with the 'PortABoxInfo' variable.

patopat, 2022-06-22:
@TomKovac, If you don't find the activation/deacrivation state you consider it disabled ?

What are corresponding hex number in the struct type ?

patopat, 2022-06-22:
@TomKovac After inspection of the DUTs generated fies, the channels are missing 
[image]
[image]

TomKovac, 2022-06-22:
Hi @patopat.

This is OK. This is just the structure for the EK1200 itself and it is empty as the EK1200 has no PDO nor InfoData etc.
There should be another structure in the Topology folder that extends from EK1200_013D0B49 and there should be the structures for boxes.

TomKovac, 2022-06-22:
> 
Hi @patopat The slave is added if:
The check box mentioned is unchecked or 'Disabled' property is unspecified or 'Disabled' property is false.
So the only combination when the slave structure should not be generated is as follows:
[image]
+ 
	<Box Id="25" **Disabled="true"** BoxType="9099" BoxFlags="#x00000020">
		<Name>__FILENAME__</Name>
		<ImageId>1000</ImageId>
		...
	</Box>

patopat, 2022-06-22:
@TomKovac  strange there is something but empty
[image]
[image]
[image]
[image]

TomKovac, 2022-06-22:
Hi @patopat, I got the same result when I manually disable all terminals after EK1200. So it seems, that the iXlinker considers somehow the terminals as disabled. Can you please check the .xti file for the first terminal, what does the line starting with <Box looks like?

patopat, 2022-06-22:
@TomKovac 
```XML
<?xml version="1.0"?>
<TcSmItem xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://www.beckhoff.com/schemas/2012/07/TcSmProject" TcSmVersion="1.0" TcVersion="3.1.4024.25" ClassName="CFlbTermDef" SubType="9099">
	<Box Id="15" BoxType="9099">
		<Name>__FILENAME__</Name>
		<ImageId>7</ImageId>
		<EtherCAT SlaveType="1" PdiType="#x0104" CycleMBoxPollingTime="0" VendorId="#x00000002" ProductCode="#x03f03052" RevisionNo="#x00120000" RepeatSupport="true" PortPhys="51" MaxSlotCount="256" MaxSlotGroupCount="1" SlotPdoIncrement="1" SlotIndexIncrement="16" Type="EL1008 8Ch. Dig. Input 24V, 3ms" Desc="EL1008">
			<SyncMan>001001000000010004000000000000000100001000010000</SyncMan>
			<Fmmu>0000000000000000001000010100000002000000000000000000000000000000</Fmmu>
			<Pdo Name="Channel 1" Index="#x1a00" Flags="#x0011" SyncMan="0">
				<Entry Name="Input" Index="#x6000" Sub="#x01">
					<Type>BIT</Type>
				</Entry>
			</Pdo>
			<Pdo Name="Channel 2" Index="#x1a01" Flags="#x0011" SyncMan="0">
				<Entry Name="Input" Index="#x6010" Sub="#x01">
					<Type>BIT</Type>
				</Entry>
			</Pdo>
			<Pdo Name="Channel 3" Index="#x1a02" Flags="#x0011" SyncMan="0">
				<Entry Name="Input" Index="#x6020" Sub="#x01">
					<Type>BIT</Type>
				</Entry>
			</Pdo>
			<Pdo Name="Channel 4" Index="#x1a03" Flags="#x0011" SyncMan="0">
				<Entry Name="Input" Index="#x6030" Sub="#x01">
					<Type>BIT</Type>
				</Entry>
			</Pdo>
			<Pdo Name="Channel 5" Index="#x1a04" Flags="#x0011" SyncMan="0">
				<Entry Name="Input" Index="#x6040" Sub="#x01">
					<Type>BIT</Type>
				</Entry>
			</Pdo>
			<Pdo Name="Channel 6" Index="#x1a05" Flags="#x0011" SyncMan="0">
				<Entry Name="Input" Index="#x6050" Sub="#x01">
					<Type>BIT</Type>
				</Entry>
			</Pdo>
			<Pdo Name="Channel 7" Index="#x1a06" Flags="#x0011" SyncMan="0">
				<Entry Name="Input" Index="#x6060" Sub="#x01">
					<Type>BIT</Type>
				</Entry>
			</Pdo>
			<Pdo Name="Channel 8" Index="#x1a07" Flags="#x0011" SyncMan="0">
				<Entry Name="Input" Index="#x6070" Sub="#x01">
					<Type>BIT</Type>
				</Entry>
			</Pdo>
		</EtherCAT>
	</Box>
</TcSmItem>
```

TomKovac, 2022-06-22:
Hi @patopat.
It is the same on my side. The only difference is the version. Mine is TcVersion="3.1.4024.22". 
I will try to update it if it comes with the new version.

TomKovac, 2022-06-22:
Hi @patopat .
I have tried it on the 3.1.4024.25 so as 3.1.4024.29. On my side, it works for both versions. For the EtherCAT_2 everything has been generated successfully or there are also some issues?  Does it even work if you just add into the empty project just one EtherCAT master, one EK1200, and one EL1008?

patopat, 2022-06-22:
@TomKovac  I have the same result. I created a testcase project

[testcase.zip](https://github.com/Inxton/iXlinker/files/8960308/testcase.zip)

TomKovac, 2022-06-23:
Hi @patopat,
I opened your solution (in 3.1.4024.29), and the PLC project was disabled. I have enabled it, run the iXlinker, and everything worked fine. When I opened the solution it was loaded with project variant PROD. Also when I just unzip your solution, without any other actions, the PROD is the active project variant. Anyway the "Disabled like" behavior could be caused by variant management. Maybe you should try to remove it or enable the ETHERCAT_1 for all variants, just to discover if this is the source of the strange behavior.  On my side it has created the proper structure even in the TEST variant with unchecked **Do not generate mapping for disabled IO** in the iXlinker.

patopat, 2022-06-23:
@TomKovac 

I removed the project variant but the struct is empty. Could you try the attached one ?

PS: I also noticed some strange behavior with project variants. It may have a bug with TwinCAT and project variants. Sometimes the PLC or I/O is activated with a specific variant. When we close the project and re-open it, the PLC or I/O is disabled for the specified variants... Few month ago, I opened a ticket in our local Beckhoff office but it has not yet replied....

[testcase02.zip](https://github.com/Inxton/iXlinker/files/8964653/testcase02.zip)

TomKovac, 2022-06-23:
Hi @patopat, I am going to check it, anyway I found in the testcase02.zip under **..\testcase02\_Vortex\iXlinker\iXlinkerLog.txt** such an error message:
**2022-06-23 08:22:41.149 +02:00 [ERR] Unable to find xti file for the box: TEST_01DI1!!!
File: C:\sw\wks\tc3\TwinCAT Project13\TwinCAT Project13\_Config\IO\ETHERCAT_1\TEST\TEST_01DI1.xti not found!!!**
It seems that it is due to a dot inside the folder name **wks.tc3**, as the start argument for the iXlinker.exe looks like this:
**-t C:\sw\wks.tc3\TwinCAT Project13\TwinCAT Project13\TwinCAT Project13.tsproj**

TomKovac, 2022-06-23:
@patopat 
I have found such a line of code also with a comment:
**                string fileName = Path.Combine(path.Replace(".","\\"), _box.File);//TODO check if dot replacement neccesarry**
🤣🤣🤣

TomKovac, 2022-06-23:
Hi @ patopat,
new version released.
https://www.nuget.org/packages/Inxton.iXlinker/0.1.4-alpha.82
This should solve the issue.

patopat, 2022-06-23:
@TomKovac  Yes it works. I also tried to the original project and I have to uncheck the option in order to generate the disabled IO.
Would you like to investigate more with me to find the root cause because all master ETHERCAT are enabled and the tool find only one ?

patopat, 2022-06-23:
@TomKovac No if I leave the option checked it finds only one master. It seems that the tool find one master disabled.

patopat, 2022-06-23:
@TomKovac   Does the USB remote is supported ? it's a Beckhoff touch panel.

TomKovac, 2022-06-23:
@patopat 
Maybe, please check the iXlinkerLog.
If you have all devices in the independent project file, they should be listed in the log.
Like this:
[INF] Project contains 1 IoDevices in individual project files
	ETHERCAT_1
Also in the case of disabled devices.

TomKovac, 2022-06-23:
> @TomKovac Does the USB remote is supported ? it's a Beckhoff touch panel.

@patopat No, just EtherCAT master. The motion is scheduled, but I am still waiting for some information from Beckhoff.

patopat, 2022-06-23:
@TomKovac  Yes I get it

2022-06-23 13:41:03.989 +02:00 [INF] Project contains 3 IoDevices in individual project files
	ETHERCAT_1
	ETHERCAT_2
	Device 1 (USB Remote)

TomKovac, 2022-06-23:
@patopat  
OK, are there also any errors in the log file like "[ERR] Unable to find xti file for the box: TEST_01DI1!!!" ?

patopat, 2022-06-23:
@TomKovac No all is fine in the log file 

```CMD
2022-06-23 13:41:03.319 +02:00 [INF] iXlinker started
2022-06-23 13:41:03.326 +02:00 [INF] with arguments:
2022-06-23 13:41:03.327 +02:00 [INF] -t C:\sw\wks.tc3\xxxxxxxxx\xxxxx.tsproj -p Debug|TwinCAT RT (x64) -c C:\sw\wks.tc3\xxxxxxxxx\xxxxx.plcproj -g True -d C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\Common7\IDE\devenv.com -n 0
2022-06-23 13:41:03.989 +02:00 [INF] Project contains 3 IoDevices in individual project files
	ETHERCAT_1
	ETHERCAT_2
	Device 1 (USB Remote)

2022-06-23 13:41:04.033 +02:00 [ERR] Unable to discover default resolution for the Plc library: Include: VisuDialogs | Namespace: VisuDialogs | DefaultResolution: null
Variable: ProjectItemGroupPlaceholderReference.DefaultResolution has an invalid reference!!!
Library is going to be ignored.
Method:GetPlcLibraries
2022-06-23 13:41:04.070 +02:00 [INF] Reading IO devices in the XAE project: "XaeXXXXX"!!!
2022-06-23 13:41:04.630 +02:00 [INF] PdoEntryStructures :0
2022-06-23 13:41:04.633 +02:00 [INF] PdoStructures :9
2022-06-23 13:41:04.633 +02:00 [INF] BoxStructures :1
2022-06-23 13:41:04.633 +02:00 [INF] DeviceStructures :3
2022-06-23 13:41:04.633 +02:00 [INF] TopologyStructures :1
2022-06-23 13:41:04.634 +02:00 [INF] Cleaning existing GVL!
2022-06-23 13:41:04.635 +02:00 [INF] Existing GVL cleared!
2022-06-23 13:41:04.635 +02:00 [INF] Cleaning existing DUTs!
2022-06-23 13:41:04.643 +02:00 [INF] Existing DUTs cleared!
2022-06-23 13:41:04.644 +02:00 [INF] Exporting Pdo entry structures to the folder C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\PdoEntries
2022-06-23 13:41:04.644 +02:00 [INF] PDO entry structures exported!!!
2022-06-23 13:41:04.644 +02:00 [INF] Adding PDO entry structures into the PLC project!!!
2022-06-23 13:41:04.645 +02:00 [INF] PDO entry structures added into the PLC project!!!
2022-06-23 13:41:04.645 +02:00 [INF] Exporting Pdo structures to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\PDOs"
2022-06-23 13:41:04.648 +02:00 [INF] PDO structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\PDOs" !!!
2022-06-23 13:41:04.650 +02:00 [INF] Adding PDO structures into the PLC project!!!
2022-06-23 13:41:04.650 +02:00 [INF] PDO structures added into the PLC project!!!
2022-06-23 13:41:04.651 +02:00 [INF] Exporting Box structures to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\Boxes"
2022-06-23 13:41:04.652 +02:00 [INF] Box structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\Boxes" !!!
2022-06-23 13:41:04.652 +02:00 [INF] Adding Box structures into the PLC project!!!
2022-06-23 13:41:04.653 +02:00 [INF] Box structures added into the PLC project!!!
2022-06-23 13:41:04.653 +02:00 [INF] Exporting Device structures to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\Devices"
2022-06-23 13:41:04.655 +02:00 [INF] Device structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\Devices" !!!
2022-06-23 13:41:04.656 +02:00 [INF] Adding Device structures into the PLC project!!!
2022-06-23 13:41:04.656 +02:00 [INF] Device structures added into the PLC project!!!
2022-06-23 13:41:04.657 +02:00 [INF] Exporting Topology structures to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\Topology"
2022-06-23 13:41:04.661 +02:00 [INF] Topology structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\Topology" !!!
2022-06-23 13:41:04.661 +02:00 [INF] Adding Topology structures into the PLC project!!!
2022-06-23 13:41:04.662 +02:00 [INF] Topology structures added into the PLC project!!!
2022-06-23 13:41:04.662 +02:00 [INF] Exporting Base structures to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\Base"
2022-06-23 13:41:04.665 +02:00 [INF] Base structures exported to the folder "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\DUTs\IO\Base" !!!
2022-06-23 13:41:04.667 +02:00 [INF] Adding Base structures into the PLC project!!!
2022-06-23 13:41:04.668 +02:00 [INF] Base structures added into the PLC project!!!
2022-06-23 13:41:04.669 +02:00 [INF] Exporting GVL : GVL_iXlinker to the folder C:\sw\wks.tc3\xxxxxxxxx\xxxxx\GVLs
2022-06-23 13:41:04.670 +02:00 [INF] GVL GVL_iXlinker exported to the folder C:\sw\wks.tc3\xxxxxxxxx\xxxxx\GVLs
2022-06-23 13:41:04.671 +02:00 [INF] Adding GVL into the PLC project!!!
2022-06-23 13:41:04.672 +02:00 [INF] GVL added into the PLC project!
2022-06-23 13:41:04.673 +02:00 [INF] Writing changes of the PLC project XXXX into the file XXXXX.plcproj!!!
2022-06-23 13:41:04.683 +02:00 [INF] Changes of the PLC project XXXXXX written into the file XXXXX.plcproj!!!
2022-06-23 13:41:04.684 +02:00 [INF] Starting process Filename: "C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\Common7\IDE\devenv.com", Arguments: ""C:\sw\wks.tc3\xxxxxxxxx\xxxxx.tsproj" /build "Debug|TwinCAT RT (x64)" /Out "C:\sw\wks.tc3\xxxxxxxxx\xxxxx\\build.txt""
2022-06-23 13:41:04.685 +02:00 [INF] Building XAE project: "C:\sw\wks.tc3\xxxxxxxxx\xxxxx.tsproj" !!!
```

TomKovac, 2022-06-23:
@patopat 
I will try to add the list of disabled devices to the log file and distinguish if it is manually disabled or disabled by project variant.

patopat, 2022-06-23:
@TomKovac  Here it's clear that the ETHERCAT_1 and ETHERCAT_2 is enabled when PROD variant is selected.
[image]
 and below when the IO are disabled when SIMU variant is selected
[image]

TomKovac, 2022-06-23:
Hi @patopat .
I have added the logging of the 'manually' disabled devices and boxes into the iXlinkerLog.txt.
Also all devices should be listed in the file: _Vortex\iXlinker\Devices.txt 
and all boxes of the device in the: _Vortex\iXlinker\DeviceLists\deviceName.txt 
Try to update:
https://www.nuget.org/packages/Inxton.iXlinker/0.1.4-alpha.85
and please let me know.

patopat, 2022-06-23:
@TomKovac  Thanks to your help. I found the issue.

I use attribute TcLinkTo to link IO variables according to project variant.
[image]

Sometimes the following window appears, and I guess when we click on Yes button 
[image]
the IO link becomes "hard link" with SIMU mapping and when I switch to PROD project variant the current item is also linked with a disabled variant. Thus the item is read as disabled even if it shows as enabled. 😊 I hope I was clear in my description

So after clearing all "hard links" it's working.

TomKovac, 2022-06-24:
Hi @patopat, I am glad that it works. Thank you for your contribution and effort.

closed 2022-06-24 by TomKovac