Inxton/iXlinker #40: In an existing project iXlinker creates the gvl but not link the variables to the related channel

LorenzoMusco, 2023-01-13:
Hello everyone,

I'm trying to use iXlinker on my TwinCAT 3 project but i noticed something strange:

i'm using a set of I/O devices such as EL2624, EL1004 or EL3174 and other ethercat slave. Due to the amount of variable that i have to create and manually link to the I/O channel, i'm trying to use iXlinker to simplfy my work. 

I noticed that if i try to use iXlinker on a new project (that still use the same set of I/O listed above) then i met no issue, iXlinker creates a new GVL and provides to link all the variable to the related channel (extremely powerful tool). 

Things change when i try to use iXlinker on an already existing project (even if it implments the same set of I/O on the same PLC). 
In this second case iXlinker is able to create the GVL with all the variable that i need but does not link them on the correct channel. 

Is there anything i can do to solve this problem?

PTKu, 2023-01-13:
Hi @LorenzoMusco, thanks for reaching out. Could you share the project with us so that we can reproduce the issue? Thanks.

TomKovac, 2023-01-17:
Hi @LorenzoMusco, please check if the PLC project is correctly built and saved before starting the iXlinker. I suppose that iXlinker creates the links, but after reopening the solution and rebuilding it, they disappear as they are not valid. So as in the subfolder _Vortex\iXlinker inside your solution folder, you should find the iXlinkerLog.txt file to discover more details. Thanks

LorenzoMusco, 2023-01-17:
> Hi @LorenzoMusco, thanks for reaching out. Could you share the project with us so that we can reproduce the issue? Thanks.

I can share the project with you, may i ask you only if i can share the project in a more private way?

TomKovac, 2023-01-17:
Hi @LorenzoMusco, you can send it to tomas.kovac@inxton.com.
Thanks